pcbreflux/espressif #1: ESP-IDF version

JanJansen47, 2017-02-04:
Hi,

My main interest is the MQTT example you made and  got compilation errors in eclipse as well as with the ESP-IDF(same errors) .  Furthermore compiled most of your apps and found that not all of them compiled correctly.  So I wonder whether you compiled them against the current ( Beta) ESP-IDF?

pcbreflux, 2017-02-04:
Hi,

sorry for that, it is a problem for me as well because the esp-idf is updated nearly every day. I try to use git pull once a week to stay updated with my last code but don't have time to update everything. So if you just focus on mqtt you could use todays (b8e2edc9) git version with xtensa 5.2.0 compiler. If you have an specific issue with one example i can manage to update this code.

have a nice day
pcbreflux

pcbreflux, 2017-02-09:
Hi,
have you checked your xtensa version? I would close this issue in some days, if its OK for you?
If not please open an issue for a specific example.

thanks and have a nice day
pcbreflux

JanJansen47, 2017-02-09:
Hi,

Not yet, I will take time in the coming weekend,

Thks,

Jan Jansen
Sagittalaan 22
5632 AL Eindhoven
+31(0)6  888  06529

JanJansen47, 2017-02-09:
Good afternoon,

Before I do a new test this weekend (also with your new mqtt) I confirm I was testing 5.2.0

Regards

Jan Jansen
Sagittalaan 22
5632 AL Eindhoven
+31(0)6  888  06529

JanJansen47, 2017-02-11:
I'm using compiler version 5.2.0.  Made a fresh download and I'm getting following from the compiler (for both publish and scribe examples)

CC cert.o
CC mqtt_subscribe_main.o
CC MQTTClient-C/src/MQTTClient.o
CC MQTTClient-C/src/mbedtls/MQTTmbedtls.o
/home/jan/ESP32_Prog/espressif-master/esp32/app/ESP32_mqtt_secure_subscribe/main/MQTTClient-C/src/mbedtls/MQTTmbedtls.c:632:1: fatal error: opening dependency file MQTTClient-C/src/mbedtls/MQTTmbedtls.d: No such file or directory
 }
 ^
compilation terminated.
make[1]: *** [MQTTClient-C/src/mbedtls/MQTTmbedtls.o] Error 1
make: *** [main-build] Error 2
jan@jan-MS-7798:~/ESP32_Prog/espressif-master/esp32/app/ESP32_mqtt_secure_subscribe$

ble_app_ibeacon:  compilation is oke
ESP32_ssl_gmail: following compilation error:
home/jan/ESP32_Prog/espressif-master/esp32/app/ESP32_ssl_gmail/main/./esp32-hal-i2c.c: In function 'i2cSetFrequency':
/home/jan/ESP32_Prog/espressif-master/esp32/app/ESP32_ssl_gmail/main/./esp32-hal-i2c.c:330:29: error: 'volatile union <anonymous>' has no member named 'scl_low_period'
     i2c->dev->scl_low_period.scl_low_period = period;
                             ^
/home/jan/ESP32_Prog/espressif-master/esp32/app/ESP32_ssl_gmail/main/./esp32-hal-i2c.c: In function 'i2cGetFrequency':
/home/jan/ESP32_Prog/espressif-master/esp32/app/ESP32_ssl_gmail/main/./esp32-hal-i2c.c:351:50: error: 'volatile union <anonymous>' has no member named 'scl_low_period'
     return APB_CLK_FREQ/(i2c->dev->scl_low_period.scl_low_period+i2c->dev->scl_
                                                  ^
/home/jan/ESP32_Prog/espressif-master/esp32/app/ESP32_ssl_gmail/main/./esp32-hal-i2c.c:352:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
make[1]: *** [esp32-hal-i2c.o] Error 1
make: *** [main-build] Error 2

Normally I use eclipse but as it takes more I stuck to the espressif IDE.

I will have a look at the code maybe there is a typo.

Jan Jansen
Sagittalaan 22
5632 AL Eindhoven
+31(0)6  888  06529

pcbreflux, 2017-02-12:
For the mqtt examples I have to guesd: can you reduce the path, maybe it's to long for yor OS?
For the esp-hal I have to change the code since espressif have done some major changes on thi, comming soon.

pcbreflux, 2017-02-12:
Hi,

this is what I am doing, to build fresh copy from git with your path:

mkdir ESP32_Prog
cd ESP32_Prog
git clone git@github.com:pcbreflux/espressif.git espressif-master
cd ~/ESP32_Prog/espressif-master/esp32/app/ESP32_mqtt_secure_subscribe
make

-> no error (only two warnings)

cd ~/ESP32_Prog/espressif-master/esp32/app/ESP32_mqtt_secure_publish
make menuconfig
make -j8 all

-> no error (only two warnings)

This is what I am using
uname -a
    Linux 4.4.0-62-generic
make -v
    GNU Make 4.1
xtensa-esp32-elf-cc --version
    xtensa-esp32-elf-cc (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 5.2.0
python -V
    Python 2.7.12

Hope this helps in a way?

Have a nice day
pcbreflux

JanJansen47, 2017-02-13:
Hello,

Same results,

You can remove my " incident"  I'm going to check the tools you use and go from there. Sorry for my interruption..  (my C knowledge is ancient , but have Java expertise)

Regards

Dr. Ir Jan Jansen
Sagittalaan 22
5632 AL Eindhoven
+31(0)6  888  06529

git clone git@github.com<mailto:git@github.com>:pcbreflux/espressif.git espressif-master

pcbreflux, 2017-02-15:
Hello,

now I've managed to update all my code to work with the latest esp-idf and xtensa 5.2.0 so I close this issue today.

have a great day
pcbreflux

closed 2017-02-15 by pcbreflux