pcbreflux/espressif #2: cant connect esp32 to cloudmqtt

HannesRammer, 2017-02-24:
Hi, i am trying your esp32 cloudmqtt examples. but i get an error when connection

hannes@hannes-VirtualBox:~/esp32/ESP32_mqtt_secure_subscribe$ make flash monitor
CC mqtt_subscribe_main.o
AR libmain.a
LD mqtt-secure-publish.elf
esptool.py v2.0-beta1
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.0-beta1
Connecting.........................
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Attaching SPI flash...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 9504 bytes to 5647...
Wrote 9504 bytes (5647 compressed) at 0x00001000 in 0.1 seconds (effective 1060.9 kbit/s)...
Hash of data verified.
Compressed 702960 bytes to 377804...
Wrote 702960 bytes (377804 compressed) at 0x00010000 in 7.2 seconds (effective 775.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 82...
Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 6922.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2388
load:0x40078000,len:6788
load:0x40080000,len:252
entry 0x40080034
I (1090) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1091) heap_alloc_caps: At 3FFB80A0 len 00027F60 (159 KiB): DRAM
I (1101) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (1122) heap_alloc_caps: At 4009CC90 len 00003370 (12 KiB): IRAM
I (1143) cpu_start: Pro cpu up.
I (1155) cpu_start: Starting app cpu, entry point is 0x40080b3c
I (0) cpu_start: App cpu up.
I (1186) cpu_start: Pro cpu start user code
I (1456) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0
I (1913) cpu_start: Starting scheduler on PRO CPU.
I (747) cpu_start: Starting scheduler on APP CPU.
I (747) wifi: frc2_timer_task_hdl:3ffbe7d8, prio:22, stack:2048
I (757) wifi: Init lldesc rx mblock:25
I (757) wifi: Init lldesc rx ampdu len mblock:7
I (757) wifi: Init lldesc rx ampdu entry mblock:4
I (757) wifi: pp_task_hdl : 3ffcb690, prio:23, stack:8192
I (767) MQTTS: Setting WiFi configuration SSID WLAN-372855...
I (767) wifi: mode : sta (24:0a:c4:00:1d:9e)
I (897) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1547) wifi: state: init -> auth (b0)
I (1557) wifi: state: auth -> assoc (0)
I (1557) wifi: state: assoc -> run (10)
I (1597) wifi: connected with WLAN-372855, channel 1
I (2777) event: ip: 192.168.2.104, mask: 255.255.255.0, gw: 192.168.2.1
I (2777) MQTTmbedtls: Connecting to m13.cloudmqtt.com:37979...
I (2917) MQTTmbedtls: Connected.
I (3777) MQTTS: MQTTClientInit  ESP32MQTTBA83DE17
I (3777) MQTTS: MQTTConnect  ...
I (5777) MQTTS: MQTTConnect not SUCCESS: -1
I (5777) MQTTmbedtls: NetworkDisconnect
I (5787) MQTTS: Starting again!
I (5787) MQTTmbedtls: Connecting to m13.cloudmqtt.com:37979...
I (5917) MQTTmbedtls: Connected.
I (6827) MQTTS: MQTTClientInit  ESP32MQTTC2A95267
I (6827) MQTTS: MQTTConnect  ...
I (8827) MQTTS: MQTTConnect not SUCCESS: -1
I (8827) MQTTmbedtls: NetworkDisconnect
I (8837) MQTTS: Starting again!
I (8837) MQTTmbedtls: Connecting to m13.cloudmqtt.com:37979...
I (8967) MQTTmbedtls: Connected.
I (9847) MQTTS: MQTTClientInit  ESP32MQTT5F43ADE2
I (9847) MQTTS: MQTTConnect  ...
I (11557) wifi: pm start, type:0

I (11847) MQTTS: MQTTConnect not SUCCESS: -1
I (11847) MQTTmbedtls: NetworkDisconnect
I (11847) MQTTS: Starting again!
I (11857) MQTTmbedtls: Connecting to m13.cloudmqtt.com:37979...
I (11977) MQTTmbedtls: Connected.
I (12857) MQTTS: MQTTClientInit  ESP32MQTT9480CC26
I (12857) MQTTS: MQTTConnect  ...
I (14857) MQTTS: MQTTConnect not SUCCESS: -1
I (14867) MQTTmbedtls: NetworkDisconnect
I (14867) MQTTS: Starting again!
E (14867) MQTTmbedtls: mbedtls_ssl_setup returned -0x7f00

I have tried all three ports cloudmqtt tells me.. and when using websocket port I also set mqtt_websocket = 1 ...

any idea what i am doing wrong? do i need special certificates ??

pcbreflux, 2017-02-25:
Hi,

can you try this one by one (not all at once):
- n MQTTmbedtls.c line 462 set
MBEDTLS_SSL_VERIFY_REQUIRED to  MBEDTLS_SSL_VERIFY_OPTIONAL 
- use make menueconfig or set in file sdkconfig and enable CONFIG_LOG_DEFAULT_LEVEL_DEBUG 
- in MQTTmbedtls.h line 65 uncomment #define MBEDTLS_MQTT_DEBUG
- check you cloudmqtt user have the acl read and/or write and setting are ok, maybe case sensitiv

pcbreflux, 2017-03-03:
Hi,

have you done any progress yet ? 
Can you post your debug output?
The debug output between 
    (3777) MQTTS: MQTTConnect ...
-> ?
   I (5777) MQTTS: MQTTConnect not SUCCESS: -1
should reveal the problem.

beejesh, 2017-05-18:
Hi ,

I have the same problem and below is what Im getting.
Im using iot.eclipse.org:8883, How to disable certificate check?
Thanks
___________________________

[1B][0;32mI (14024) MQTTmbedtls: Connected.[1B][0m
[1B][0;33mW (16234) MQTTmbedtls: Failed to verify peer certificate![1B][0m
[1B][0;33mW (16234) MQTTmbedtls: verification info:   ! The certificate is not correctly signed by the trusted CA
[1B][0m
[1B][0;32mI (16564) MQTTmbedtls: NetworkDisconnect[1B][0m
[1B][0;32mI (16564) MQTT: MQTTClientInit  ...[1B][0m
[1B][0;32mI (16574) MQTT: MQTTClientInit  ESP32MQTT[1B][0m
[1B][0;32mI (16574) MQTT: MQTTConnect  ...[1B][0m
0x10,0x19,0x00,0x04,0x4D,0x51,0x54,0x54,0x04,0xC2,0x00,0x05,0x00,0x09,0x45,0x53,0x50,0x33,0x32,0x4D,0x51,0x54,0x54,0x00,0x00,0x00,0x00,
0x82,0x9B,0x01,0x02,0x03,0x04,0x11,0x1B,0x03,0x00,0x4C,0x53,0x57,0x50,0x05,0xC0,0x03,0x01,0x01,0x0B,0x46,0x57,0x51,0x31,0x31,0x49,0x50,0x56,0x57,0x04,0x01,0x02,0x03,
[1B][0;32mI (16594) MQTT: MQTTConnect not SUCCESS: -1[1B][0m
0xE0,0x00,
0x82,0x82,0x01,0x02,0x03,0x04,0xE1,0x02,
[1B][0;32mI (16604) MQTTmbedtls: NetworkDisconnect[1B][0m
_____________________________

jaggi, 2017-07-27:
Hi,

can you try this one by one (not all at once):

    n MQTTmbedtls.c line 462 set
    MBEDTLS_SSL_VERIFY_REQUIRED to MBEDTLS_SSL_VERIFY_OPTIONAL
    use make menueconfig or set in file sdkconfig and enable CONFIG_LOG_DEFAULT_LEVEL_DEBUG
    in MQTTmbedtls.h line 65 uncomment #define MBEDTLS_MQTT_DEBUG
    check you cloudmqtt user have the acl read and/or write and setting are ok, maybe case sensitiv
Reply to this solution,

OPTIONS: I18n 
Compiled on Jan  1 2014, 17:13:19.
Port /dev/ttyUSB0, 10:13:52

Press CTRL-A Z for help on special keys

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3420
ho 0 tail 12 room 4
load:0x40078000,len:10384
load:0x40080000,len:252
entry 0x40080034
I (2157) wifi: wifi firmware version: 72ddf26
I (2157) wifi: config NVS flash: enabled
I (2157) wifi: config nano formating: disabled
I (2177) wifi: Init dynamic tx buffer num: 32
I (2177) wifi: Init dynamic rx buffer num: 32
I (2177) wifi: wifi driver task: 3ffbd7f4, prio:23, stack:4096
I (2177) wifi: Init static rx buffer num: 10
I (2187) wifi: Init dynamic rx buffer num: 32
I (2187) wifi: Init rx ampdu len mblock:7
I (2187) wifi: Init lldesc rx ampdu entry mblock:4
I (2197) wifi: wifi power manager task: 0x3ffc2bbc prio: 21 stack: 2560
I (2207) MQTTS: Setting WiFi configuration SSID Redmi...
I (2207) wifi: wifi timer task: 3ffc3c3c, prio:22, stack:3584
I (2247) wifi: mode : sta (24:0a:c4:05:a8:84)
I (2367) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (3017) wifi: state: init -> auth (b0)
I (4017) wifi: state: auth -> init (2)
I (4027) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (4147) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (4147) wifi: state: init -> auth (b0)
I (4147) wifi: state: auth -> assoc (0)
I (4167) wifi: state: assoc -> run (10)
I (14177) wifi: pm start, type:0

I (14177) wifi: state: run -> init (2)
I (14177) wifi: pm stop, total sleep time: 0/688

I (14177) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (14297) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (14297) wifi: state: init -> auth (b0)
I (14307) wifi: state: auth -> assoc (0)
I (14327) wifi: state: assoc -> run (10)
I (24327) wifi: pm start, type:0

I (24327) wifi: state: run -> init (2)
I (24327) wifi: pm stop, total sleep time: 0/649

I (24327) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (24457) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (24457) wifi: state: init -> auth (b0)
I (24457) wifi: state: auth -> assoc (0)
I (24477) wifi: state: assoc -> run (10)
I (34477) wifi: pm start, type:0

I (34477) wifi: state: run -> init (2)
I (34477) wifi: pm stop, total sleep time: 0/650

I (34477) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (34607) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (34607) wifi: state: init -> auth (b0)
I (34607) wifi: state: auth -> assoc (0)
I (34627) wifi: state: assoc -> run (10)
I (44627) wifi: pm start, type:0

I (44627) wifi: state: run -> init (2)
I (44627) wifi: pm stop, total sleep time: 0/639

I (44627) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (44757) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (44757) wifi: state: init -> auth (b0)
I (44757) wifi: state: auth -> assoc (0)
I (44777) wifi: state: assoc -> run (10)
I (54777) wifi: pm start, type:0

I (54777) wifi: state: run -> init (2)
I (54777) wifi: pm stop, total sleep time: 0/646

I (54777) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (54907) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (54907) wifi: state: init -> auth (b0)
I (54907) wifi: state: auth -> assoc (0)
I (54927) wifi: state: assoc -> run (10)
I (64927) wifi: pm start, type:0