pcbreflux/espressif #3: port this to esp8266 sdk and or sming framework

zenonasz, 2017-03-01:
Hi,

i am going to try this anyway, but i was wondering if you can point me to dependencies that might be missing before trying to port this to esp8266 nonos sdk?

Looking for a client that supports mqtt over websockets for the esp8266 nonos sdk

Regards,

Zenon

pcbreflux, 2017-03-12:
Hi, sorry but my answer seems to bee sendet to /dev/null. So this the context I have written.
I use the MQTT python client as an insperation for websockets so please have a look. I have do nearly nothing with python but is easy to read with a C/C++ background. https://github.com/eclipse/paho.mqtt.python

pcbreflux, 2017-03-12:
Hi, 

the MbedTLS can be a challang on the esp8266 so see here for a not recommended but unencrypted implementation https://github.com/nkolban/esp32-snippets/tree/master/networking/mqtt/paho_mqtt_embedded_c

have a nice day
pcbreflux

zenonasz, 2017-03-12:
Hi,

thanks for the reply and the resources, we have somehow figure out the flow and we already did a proof of concept just by reading your code. We did that using the sming MQTT and Websockets libraries.

If you dont mind, i can keep this post active for questions.

Thanks

pcbreflux, 2017-03-13:
Hi,

shure and sorry for the delay. Was shure I write just a day after your post.

Thanks.