pcbreflux/espressif #4: BLUFI_DEMO: app_main init bluedroid failed

neilyoung, 2017-03-12:
ESP-32 IDF fresh install from git. Toolchain OK. Make works. After flashing this on a Sparkfun:

I (263) boot: ESP-IDF v2.0-rc1-257-g4745895 2nd stage bootloader
I (264) boot: compile time 16:45:51
I (291) boot: Enabling RNG early entropy source...
I (291) boot: SPI Speed      : 40MHz
I (298) boot: SPI Mode       : DIO
I (310) boot: SPI Flash Size : 4MB
I (323) boot: Partition Table:
I (334) boot: ## Label            Usage          Type ST Offset   Length
I (357) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (380) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (403) boot:  2 factory          factory app      00 00 00010000 00100000
I (427) boot: End of partition table
I (440) boot: Disabling RNG early entropy source...
I (457) boot: Loading app partition at offset 00010000
I (2131) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512)
I (2131) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x3e5c0 (255424) map
I (2148) boot: segment 2: paddr=0x0005e5d0 vaddr=0x3ffc0000 size=0x02754 ( 10068) load
I (2179) boot: segment 3: paddr=0x00060d2c vaddr=0x40080000 size=0x00400 (  1024) load
I (2202) boot: segment 4: paddr=0x00061134 vaddr=0x40080400 size=0x1ab60 (109408) load
I (2280) boot: segment 5: paddr=0x0007bc9c vaddr=0x400c0000 size=0x00034 (    52) load
I (2281) boot: segment 6: paddr=0x0007bcd8 vaddr=0x00000000 size=0x04330 ( 17200)
I (2297) boot: segment 7: paddr=0x00080010 vaddr=0x400d0018 size=0xb1714 (726804) map
I (2323) boot: segment 8: paddr=0x0013172c vaddr=0x50000000 size=0x00004 (     4) load
I (2352) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (2373) heap_alloc_caps: At 3FFD1750 len 0000E8B0 (58 KiB): DRAM
I (2394) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (2416) heap_alloc_caps: At 4009AF60 len 000050A0 (20 KiB): IRAM
I (2437) cpu_start: Pro cpu up.
I (2448) cpu_start: Starting app cpu, entry point is 0x40080b90
I (0) cpu_start: App cpu up.
I (2480) cpu_start: Pro cpu start user code
I (2541) cpu_start: Starting scheduler on PRO CPU.
I (81) cpu_start: Starting scheduler on APP CPU.
I (81) wifi: wifi firmware version: 90b1b8b
I (81) wifi: config NVS flash: enabled
I (81) wifi: config nano formating: disabled
I (111) wifi: Init dynamic tx buffer num: 32
I (111) wifi: wifi driver task: 3ffd8fd4, prio:23, stack:3584
I (111) wifi: Init static rx buffer num: 10
I (111) wifi: Init dynamic rx buffer num: 0
I (111) wifi: Init rx ampdu len mblock:7
I (121) wifi: Init lldesc rx ampdu entry mblock:4
I (121) wifi: wifi power manager task: 0x3ffde37c prio: 21 stack: 2560
I (131) BLUFI_DEMO: Setting WiFi configuration xxxxxxxxxxxx
I (141) wifi: wifi timer task: 3ffdf3fc, prio:22, stack:3584
I (161) phy: phy_version: 329, Feb 22 2017, 15:58:07, 1, 0
I (161) wifi: Init ampdu: 1
I (161) wifi: mode : sta (24:0a:c4:05:98:8c)
BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6e0, len 00001830
set .bss 0x0 from 3ffb8000 to 3ffbff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:05:98:8E
NVDS MAGIC FAILED
RF Init OK with coex
E (341) BT: Conroller not initialised

E (341) BLUFI_DEMO: app_main init bluedroid failed

I (1841) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:11
I (2501) wifi: state: init -> auth (b0)
I (2501) wifi: state: auth -> assoc (0)
I (2511) wifi: state: assoc -> run (10)
I (2531) wifi: connected with HOME, channel 11
I (3581) event: ip: 192.168.178.47, mask: 255.255.255.0, gw: 192.168.178.1

Any idea what might be wrong with this?

neilyoung, 2017-03-12:
The original blufi_main.c does it, so it's probably something with the sdkconfig...

neilyoung, 2017-03-12:
Arghh. The initialization sequence has changed:

Former way:

    initialise_wifi();

    esp_bt_controller_init();

    ret = esp_bluedroid_init();

Now:

    initialise_wifi();

    esp_bt_controller_init();

    ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM);
    if (ret) {
        BLUFI_ERROR("%s enable bt controller failed\n", __func__);
        return;
    }

    ret = esp_bluedroid_init();

After esp_bt_controller_init a call to esp_bt_controller_enable is required.

pcbreflux, 2017-03-12:
Thank you, had the same problem last week with my ble gateway example. Hopfully the code will be updated this week. But hope some day the esp-idf don't update essential routines that often.