pcbreflux/espressif #5: BLUFI_MAIN: sheep counter not working for me anymore

neilyoung, 2017-03-12:
Hi,

I think your sheep counter code is not working anymore (at least it is not working for me): Although I have a bunch of RadiusNetwork iBeacons on air, the code is not scanning either one of them (see attached image from an Android scanner)

Then - even though the stack would give me an iBeacon scan, it would not pass your iBeacon detector. Please check the iBeacon advertisement of an RadBeacon. It differs from your Nordic device in the third byte.

Other than that all other beacons around are detected.

What might be wrong?

neilyoung, 2017-03-12:
Oops, sorry. Was a bit too fast (and too blind). The iBeacons are scanned. But please double check the detection routine

Regards

neilyoung, 2017-03-12:
OK, finally: You can _for sure_ forget about the iBeacon scan results (at least on a Sparkfun latest stack): Not only, that scans are just scattered, they are also complete nonsense.

Look at these two consecutive (hey, consecutive means: 30 secs in between) readings of one and the same stationary iBeacon:

BEACON_DEMO: BDA 0C:F3:EE:04:1A:A8
BEACON_DEMO: rssi: -27
BEACON_DEMO: BDA 0C:F3:EE:04:1A:A8
BEACON_DEMO: rssi: -79

This is simply impossible.

pcbreflux, 2017-03-12:
You have also find the rumours about the RSSI value, but I don't see it in the time of my development. If this is a issue, this should go to the esp-idf github. Can you open one with you findings? https://github.com/espressif/esp-idf/issues

neilyoung, 2017-03-12:
Done. Same issues with this snippet BTW https://github.com/nkolban/esp32-snippets/blob/master/BLE/scanner/ble1.c

neilyoung, 2017-03-12:
Please don't forget the iBeacon detection problem mentioned above. This _is_ your problem :)

neilyoung, 2017-03-12:
Interesting turn in this matter: Up to now I just tested with RadBeacons (Dot). Now I used my iPhone as virtual Estimote beacon. And scans come in. The RSSI values are still nonsense (e.g. -8 very close to the beacon), but at least I receive scans...