pcbreflux/espressif #7: Make on secure MQTT examples yeilds dependency file error

lakidd, 2017-03-19:
Hi
My googling on this topic yields very little useful information on this type of error.

`
<snip>
CC mqtt_subscribe_main.o
CC MQTTClient-C/src/MQTTClient.o
CC MQTTClient-C/src/mbedtls/MQTTmbedtls.o
/Users/lakidd1/esp32/pcbreflux/esp32/app/ESP32_mqtt_secure_subscribe/main/MQTTClient-C/src/mbedtls/MQTTmbedtls.c:632:1: fatal error: opening dependency file MQTTClient-C/src/mbedtls/MQTTmbedtls.d: No such file or directory
 }
 ^
compilation terminated.
make[1]: *** [MQTTClient-C/src/mbedtls/MQTTmbedtls.o] Error 1
make: *** [main-build] Error 2`

I've seen comments that paths can be too long but this is OS X so shouldn't be an issue.
Other examples build fine. 
Any thoughts ? I've tried creating that directory in the location its complaining about

pcbreflux, 2017-03-20:
Hi,

sorry still have no clue: my dependency files on ubuntu would build fine.
Like MQTTmbedtls.d under build/main/MQTTClient-C/src/mbedtls/MQTTmbedtls.d.

But maybe can you exclude the longpath error by shorten/copy  
/Users/lakidd1/esp32/pcbreflux/esp32/app/ESP32_mqtt_secure_subscribe
to something like
/Users/lakidd1/esp32/esp32_msecsub
and compile again?

    make clean
    make menuconfig
    make

Just to exclude this also, do you use a new esp-idf?

lakidd, 2017-03-20:
Ok so this gets interesting...ubuntu on 14.04 gives me exactly the same result.
If i go in and manually create the directory, it compiles which is a good start.
The question is, why is the directory creation failing in the makefile ?
I tried shortening the path with no luck

pcbreflux, 2017-03-21:
Hi, 
using Ubuntu 16.04 but I delete the build directory very often an have not discovered any problem with make and all the included scripts from espressif. Sorry, still got no clue.

dattranfiot, 2017-05-08:
Hi,
I have encountered an error like you and I edited it as follows:

COMPONENT_ADD_INCLUDEDIRS := . MQTTClient-C/src/mbedtls MQTTClient-C/src MQTTPacket/src
in file "component.mk"