pcbreflux/espressif #8: ble_ibeacon

fpereiracostaa, 2017-04-13:
BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6em, (en ) 0pu830art: Stars 0x0 cheduler o000 toC3f.[0m
, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:04:D0:BA
NVDS MAGIC FAILED
RF Init OK with coex
BLE advt task start
BLE Advertise, flag_send_avail: 0, cmd_sent: 0
BLE Advertise, flag_send_avail: 0, cmd_sent: 0
BLE Advertise, flag_send_avail: 0, cmd_sent: 0
BLE Advertise, flag_send_avail: 0, cmd_sent: 0
BLE Advertise, flag_send_avail: 0, cmd_sent: 0

Any advice on this problem? I can put the esp-idf example working and it reaches cmd_sent: 4, but with this ibeacon implementation i cant... :(

pcbreflux, 2017-04-13:
Can you tell me your esp-idf version (git log)? So maybe espressif have change something in the bt-stack and the example need to be updated?

pcbreflux, 2017-04-23:
Hi, any progress so far ?

fpereiracostaa, 2017-04-23:
Hi, thanks a lot for your reply. How can i see the esp-idf version?

pcbreflux, 2017-04-24:
Open a command line. Change to your esp-idf path (like cd $IDF_PATH) and type git log.
The fist entry is your version. 

You see something like:
commit fd3ef4cdfe1ce747ef4757205f4bb797b099c9d9
Merge: 94a61389 52c378b4
Author: Angus Gratton <angus@espressif.com>
Date:   Fri Apr 21 12:27:32 2017 +0800

then I can do this:

git clone https://github.com/espressif/esp-idf.git esp-idf
cd esp-idf
git checkout fd3ef4cdfe1ce747ef4757205f4bb797b099c9d9
git submodule update --init

And compile against your Version.

Thanks.