pcbreflux/espressif #15: Build problem

vizisens, 2017-07-31:
Hi,

I think due to version upgrade in ESP-IDF libraries, we encounter mising library module problem 

OS is Ubuntu 16.04 desktop version.

Any idea?
regards

In  **ESP32_gatts_mqtt_gateway**

`CC port/debug/lwip_debug.o
AR liblwip.a
CC gateway_main.o
/home/gokhan/work/app/ESP32_gatts_mqtt_gateway/main/./gateway_main.c:23:35: fatal error: freertos/heap_regions.h: No such file or directory
compilation terminated.
/home/gokhan/esp/esp-idf/make/component_wrapper.mk:228: recipe for target 'gateway_main.o' failed
make[1]: *** [gateway_main.o] Error 1
/home/gokhan/esp/esp-idf/make/project.mk:391: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
`

In  **ESP32_blufi_beacon_tracker**
`
CC blufi_beacon_scan.o
CC blufi_main.o
/home/gokhan/work/app/ESP32_blufi_beacon_tracker/main/./blufi_main.c: In function 'app_main':
/home/gokhan/work/app/ESP32_blufi_beacon_tracker/main/./blufi_main.c:285:5: error: too few arguments to function 'esp_bt_controller_init'
     esp_bt_controller_init();
     ^
In file included from /home/gokhan/work/app/ESP32_blufi_beacon_tracker/main/./blufi_main.c:29:0:
/home/gokhan/esp/esp-idf/components/bt/include/bt.h:150:11: note: declared here
 esp_err_t esp_bt_controller_init(esp_bt_controller_config_t *cfg);
           ^
/home/gokhan/esp/esp-idf/make/component_wrapper.mk:228: recipe for target 'blufi_main.o' failed
make[1]: *** [blufi_main.o] Error 1
/home/gokhan/esp/esp-idf/make/project.mk:391: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
`

gerisse, 2018-03-26:
i have a similar problem : i applied your method to build a project with eclipse ; all seems correct , except this error : 
./main.c:23:35: fatal error: freertos/heap_regions.h: No such file or directory
i updated esp-idf repository and its submodules. But the error comes back.
An idea ?
thanks

Trubochkin, 2018-04-02:
I also have a problem (on Windows 10, Eclipse):
make[1]: *** [/home/Oleksii/esp-idf/make/component_wrapper.mk:274: blufi_main.o] Error 1
make: *** [C:/msys32/home/Oleksii/esp-idf/make/project.mk:450: component-main-build] Error 2

St7ven, 2018-04-10:
Hi,

I get the same problem as well. ./main.c:23:35: fatal error: freertos/heap_regions.h: No such file or directory

I look the the repository and this file isn't there. So I just remove this include and the template file works well.

hiteshpratyushv, 2018-05-09:
Similar Problem for me in ESP32_getts_mqtt_gateway...

CC build/main/gatts_profile.o
C:/msys32/home/mahe/esp/ESP32_gatts_mqtt_gateway/main/gatts_profile.c:25:21: fatal error: bta_api.h: No such file or directory
compilation terminated.
make[1]: *** [/home/mahe/esp/esp-idf/make/component_wrapper.mk:286: gatts_profile.o] Error 1
make: *** [C:/msys32/home/mahe/esp/esp-idf/make/project.mk:467: component-main-build] Error 2