pcbreflux/espressif #18: mbedtls_ssl_handshake issue

weaf, 2017-08-11:
Hello, 

I am having problem with any of the mqtt examples to work on the ESP32. Here is an output from the ESP:
_
I (3824) MQTTS: Start MQTT Task ...
I (3824) MQTTS: NetworkConnect 192.168.1.90:1883 ...
I (3834) MQTTmbedtls: Connecting to 192.168.1.90:1883...
I (3844) MQTTmbedtls: Connected.
**E (3854) MQTTmbedtls: mbedtls_ssl_handshake returned -0x7280**
I (3854) MQTTmbedtls: NetworkDisconnect
I (3864) MQTTS: MQTTClientInit  ...
I (3864) MQTTS: MQTTConnect  ...
I (3864) MQTTS: MQTTConnect not SUCCESS: -1
I (3874) MQTTmbedtls: NetworkDisconnect
I (3874) MQTTS: 60...
I (13184) wifi: pm start, type:0

I (13184) wifi: active cnt: 16
_

loops this periodically. It seems the problem is with mbedtls but still after a couple of hours I haven't managed to solve it. Any clue on what i can be?

Regards!

LeoLaiFookSiang, 2017-09-24:
Hi, I meet the same issue too and I am not able to solve it, do you manage to solve it ?

weaf, 2017-09-24:
I didn't manage to solve it. This worked for me and I have implemented it in my project : https://github.com/256dpi/esp-mqtt

LeoLaiFookSiang, 2017-09-24:
Hi, I just solve the issue, what I have done is I use https://www.cloudmqtt.com/ service. After I register they provide 3 kind of port which is port, ssl port and web socket port. Use the SSL port and create user with password and set into code. After that run this command to get CA certificaate

openssl s_client -showcerts -connect m20.cloudmqtt.com:20001 (For example, the port must be the ssl port)

Copy the last cert in the chain output by the server and paste it into cert.c and it should works.

Take notes that the port must be encrypted (applied SSL) else I think you will have trouble to run it.

AmrutaCh, 2018-02-16:
This also happens if the task doesn't get enough time to execute. I have faced this and solved by reducing number of tasks and giving higher priority to mqtt task