pcbreflux/espressif #32: BLE+WiFi does't work together

Bhavikjp, 2018-04-13:
I have used "**ESP32_BLE_beacconscan.ino**" code for scanning beacons .It works perfect.
But when i added WiFi initialization code with beaconscan code, neither beaconscan work nor WiFi.
If i run individually,it works.
I wants to use **WiFi + BeaconScan** simultaneously.
Is there any way to solve out this issue?
Thanks!!

moononournation, 2018-04-13:
I have tried ESP32 works with BLE and WiFi, you may see more details at my instuctables: https://www.instructables.com/id/Nano-ESP32-BLE-Scanner/

Bhavikjp, 2018-04-13:
Thanks moononournation for quick reply!!
I have uploaded your code. Still it does't work.
I got following response.
Response:
******************************************************************************
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
user code done
*******************************************************************************
I have used ESP-Wroom-32 dev board.
How to resolve it?

Thanks!!!

moononournation, 2018-04-13:
I am also using Wroom module (same as your dev board), if you have followed all the steps in my instructables, the serial monitor should show something like that:
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
E (18) wifi: esp_wifi_stop 809 wifi is not init
ESP32 BLE Scanner
Start BLE scan for 30 seconds...

closed 2018-04-16 by Bhavikjp